IL2232-HT2022-HIL/HARDWARE_DEV #3: Aquire and familiarize with hardware boards

HolgerStenberg, 2022-09-15:
Find out about, install development environment.

HolgerStenberg, 2022-09-15:
There are two boards that will be part of the project:
- The MCU Development Board: Nucleo-L476RG
- The MCU HiL Board : Nucleo-F767ZI 

I will investigate if these boards can be found as CAD-files

HolgerStenberg, 2022-09-16:
Additional information from mattias regarding what signals that are of interest is required to continue

HolgerStenberg, 2022-09-19:
MII or RMII ehternet interface? 
- [ ] Solved?

sofiecmf, 2022-09-19:
We need for sure that these peripherals exists and do not collide with anything else :

1 USART bus
1 I2C bus
1 SPI bus
Ethernet (RMII or MII)
DAC for HiL (for potentiometer emulator)
ADC for Development MCU (for potentiometer read)

HolgerStenberg, 2022-09-19:
We are thinking of combining the power from both pcbs with only one usb cable (for ease-of-use).
Currently looking into how to configure the boards such that the MCU Development Board can power the adapter board and the HiL Board

HolgerStenberg, 2022-09-19:
On small board: 
dis-connect JP1 for 300 mA over ST-link

closed 2022-09-27 by sofiecmf