IL2232-HT2022-HIL/HARDWARE_DEV #7: Source and solder on X3 crystal oscillator on HiL-board

olleblo, 2022-09-30:
First investigate if HSI oscillator (16 MHZ ) on  board is enough and is usable.
In order to get intra-IC connectivity for ethernet communication to work properly and reliably, Holger states that we need to solder on a missing crystal oscillator (X3 on the PCB). See picture below for placement.

DONE CRITERIAS: 
- [ ] When the correct crystal is soldered on.
- [ ] Jumper SB148 and SB163 OFF (Underside)
- [ ] SB8 and SB9 soldered 
- [ ] C37 and C38 soldered with 4.3 pF capacitors. Footprint: 0603
- [ ] SB112 and SB149 OFF (Underside)
- [ ] Function has been verified

[image]

HolgerStenberg, 2022-09-30:
paused due to order being made by @matthiasthomasbecker

HolgerStenberg, 2022-09-30:
@matthiasthomasbecker Will you also source the pin headers needed for the adapter PCB? 
It should be 2x of the HiL-morpho connectors and 2x of the MCU-morpho connectors, with the other gender of course.

matthiasthomasbecker, 2022-10-04:
yes, I will provide those. I think we should have them in our local stock

matthiasthomasbecker, 2022-10-17:
Hi @HolgerStenberg , the components arrived now. If one of you is in Kista you can collect them. 
Please let me know in advance to make sure that I am there as well when you want to collect them.

HolgerStenberg, 2022-10-18:
Perfect! We will make sure someone from our team gets them from Kista. This is probably a good task for @Fkar97 !

Fkar97, 2022-10-20:
@matthiasthomasbecker I'll be in Kista tomorrow, when would be a good time?

closed 2022-11-11 by HolgerStenberg