IL2232-HT2022-HIL/HARDWARE_DEV #16: Send the adapter board for production

HolgerStenberg, 2022-10-20:
Once the board has been reviewed by matthias, it can be ordered from JLC PCB. The required files for ordering can be found under "/ADAPTER_BOARD/GERBERS/V1". Only request from project group is that the PCB will be black, should not affect cost.

HolgerStenberg, 2022-10-20:
Give me one second to fix an issue regarding the ordering. We need to verify that certain functions work on the HiL board first (may be conflicts)

HolgerStenberg, 2022-11-11:
@matthiasthomasbecker Is this completed?

matthiasthomasbecker, 2022-11-11:
Hi Holger,

No this is not completed.
I am still waiting for the updated BOM files to work with JLCPCB.

The current version does not work with their page.

Could you please send me an updated version by email once done?

Best,
Matthias

Date: Friday, 11 November 2022 at 11:52
Cc: Matthias Becker ***@***.***>, Mention ***@***.***>

HolgerStenberg, 2022-11-11:
Certainly! I will try to get it done today. 👍

closed 2022-12-15 by HolgerStenberg