IL2232-HT2022-HIL/HARDWARE_DEV #18: Project management: write status report 2

sofiecmf, 2022-11-28:
In this course, two status reports are mandatory to hand in. This will be the second one of the two status reports and its due date is on Monday the 5th of December at 23:59.

The status report should follow up on aspects from the project plan that are completed, and on any changes made. The risk report should be revisited and the time and resource plan updated.

Link to document:
*not yet created*

To do:

 

- [ ] create skeleton document in overleaf

 

- [ ] partition the workload

 

- [ ] inform team-mates on what parts need work

 

- [ ] upload first draft to github for matthias to review

 

- [ ] address any comments on updates

 

- [ ] hand in on canvas

sofiecmf, 2022-11-30:
Hi team,

I have created a document for status report 2 and made some comments where information needs to be added. These are the parts where I need your assistance in writing:

- [ ] Update progress on deliverable D2
- [ ] Update progress on deliverable D3 (@olleblo?)
- [ ] Update progress on deliverable D5
- [ ] Write about what has been done in relation to the resource plan and deliverables
- [ ] Write about a challenge (if there is one)
- [ ] Write about any alterations made to the project in relation to the project plan (if there is one)

Please also update your hours in the resource document on google drive! :)

These are the parts I have assigned for myself to write, but if you want to add something to these parts just go ahead:

- [ ] Write current status
- [ ] Update progress on deliverable D1
- [ ] Update progress on deliverable D4
- [ ] Write about hours worked and hours budgeted
- [ ] Summarize the issue with SPI 
- [ ] Update Gantt chart
- [ ] Update Resource plan

sofiecmf, 2022-12-04:
Questions:

What is deliverable D4.6, and have we made any progress on this?

HolgerStenberg, 2022-12-04:
Will update tomorrow! @sofiecmf Could you assign me to some parts?

sofiecmf, 2022-12-04:
Cool, thank you @HolgerStenberg ! Could you write some words on our progress of deliverables D2 and D5? I think you have knowledge of this :)

HolgerStenberg, 2022-12-05:
@sofiecmf I think my parts are good to go now 👍

Yuxiaowa, 2022-12-05:
Hi everyone, 
I have managed to fill in the worked time of mine, and I also added my parts in Current status description of status report 2.
I cannot think of any challenges in the third chapter.
If there is anything else that I need to add or modify, please let me know ;)

closed 2022-12-15 by HolgerStenberg